alambe94/I-CUBE-USBD-Composite #1: EP Buffers

lbthomsen, 2021-09-21:
First of all - let me apologize up front if this issue is a bit of a mess.  I am still trying to nail down exactly what is happening and this issue should probably be split into 2-3 different issues.

After the announcement of this project on Reddit (https://www.reddit.com/r/embedded/comments/pqql3v/stm32_usb_composite_class_wrapper/) I immediately tried and ran into an issue of no enumeration (trying a composite with cdc, hid and msc).  On Reddit it was suggested to change `usbd_conf.c`:

```
HAL_PCDEx_SetRxFiFo(hpcd_USB_OTG_PTR, 1024); // ALL OUT EP Buffer
```

To: 

```
HAL_PCDEx_SetRxFiFo(hpcd_USB_OTG_PTR, 128); // ALL OUT EP Buffer
```

Which made the device enumerate.

These experiments were all done on a regular "black pill" board with stm32f411ce mcu.  According to the reference manual that MCU has got "1.25 kB" of dedicated USB FIFO buffer memory (presumably that is 1280 byte or 0x0500), so it is quite obvious why the change suggested is necessary (at least on this particular MCU).

However, even _with_ this change there are still some issues.  trying the composite cdc, hid, msc and changing the fifo values to:

```
    HAL_PCDEx_SetRxFiFo(hpcd_USB_OTG_PTR, 0x80); // ALL OUT EP Buffer
    HAL_PCDEx_SetTxFiFo(hpcd_USB_OTG_PTR, 0, 0x40); // EP0 IN
    for (uint8_t i = 0; i < USBD_CDC_ACM_COUNT; i++)
    {
      HAL_PCDEx_SetTxFiFo(hpcd_USB_OTG_PTR, (CDC_IN_EP[i] & 0x7F), 0x80);
      HAL_PCDEx_SetTxFiFo(hpcd_USB_OTG_PTR, (CDC_CMD_EP[i] & 0x7F), 0x40);
    }
    HAL_PCDEx_SetTxFiFo(hpcd_USB_OTG_PTR, (CUSTOM_HID_IN_EP & 0x7F), 0x80);
    HAL_PCDEx_SetTxFiFo(hpcd_USB_OTG_PTR, (MSC_IN_EP & 0x7F), 0x80);
```

Enumerates like this:

```
[334523.568812] usb 2-4.5.4: new full-speed USB device number 122 using xhci_hcd
[334523.670669] usb 2-4.5.4: New USB device found, idVendor=0483, idProduct=52a4, bcdDevice= 2.00
[334523.670671] usb 2-4.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[334523.670672] usb 2-4.5.4: Product: STM32 COMPOSITE DEVICE
[334523.670673] usb 2-4.5.4: Manufacturer: STMicroelectronics
[334523.670673] usb 2-4.5.4: SerialNumber: 318A34583030
[334523.688054] cdc_acm 2-4.5.4:1.0: ttyACM1: USB ACM device
[334523.689281] hid-generic 0003:0483:52A4.001B: hiddev3,hidraw6: USB HID v1.11 Device [STMicroelectronics STM32 COMPOSITE DEVICE] on usb-0000:00:14.0-4.5.4/input2
[334523.689509] usb-storage 2-4.5.4:1.3: USB Mass Storage device detected
[334523.689641] scsi host7: usb-storage 2-4.5.4:1.3
[334524.777016] usb 2-4.5.4: reset full-speed USB device number 122 using xhci_hcd
[334524.878479] cdc_acm 2-4.5.4:1.0: ttyACM1: USB ACM device
[334524.973015] usb 2-4.5.4: reset full-speed USB device number 122 using xhci_hcd
[334525.074569] cdc_acm 2-4.5.4:1.0: ttyACM1: USB ACM device
[334525.173023] usb 2-4.5.4: reset full-speed USB device number 122 using xhci_hcd
[334525.274489] cdc_acm 2-4.5.4:1.0: ttyACM1: USB ACM device
[334525.373025] usb 2-4.5.4: reset full-speed USB device number 122 using xhci_hcd
[334525.474606] cdc_acm 2-4.5.4:1.0: ttyACM1: USB ACM device
```

This experiment is here: https://github.com/lbthomsen/blackpill/tree/master/comptest

I have not yet tested the HID, but oddities are as follows:

The CDC resets 4 times (as seen above).  This does not happen when using ST's own HAL CDC.  It does however work (at lest for transmitting).  Using:

```
int _write(int file, char *ptr, int len) {
    CDC_Transmit(0, (uint8_t *)ptr, len);
    return len;
}
```

works - serial data is received on `/dev/ttyACMx`.

The mass storage device however doesn't register as a storage device.  Using ST's MSC driver (example here: https://github.com/lbthomsen/blackpill/tree/master/msc), enumerates like this:

```
[335036.092341] usb 2-4.5.4: new full-speed USB device number 123 using xhci_hcd
[335036.202003] usb 2-4.5.4: New USB device found, idVendor=0483, idProduct=572a, bcdDevice= 2.00
[335036.202005] usb 2-4.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[335036.202006] usb 2-4.5.4: Product: STM32 Mass Storage
[335036.202007] usb 2-4.5.4: Manufacturer: STMicroelectronics
[335036.202008] usb 2-4.5.4: SerialNumber: 318A34583030
[335036.208286] usb-storage 2-4.5.4:1.0: USB Mass Storage device detected
[335036.208686] scsi host7: usb-storage 2-4.5.4:1.0
[335037.240855] scsi 7:0:0:0: Direct-Access     STM      Product          0.01 PQ: 0 ANSI: 2
[335037.241056] sd 7:0:0:0: Attached scsi generic sg4 type 0
[335037.242979] sd 7:0:0:0: [sde] 65536 512-byte logical blocks: (33.6 MB/32.0 MiB)
[335037.243191] sd 7:0:0:0: [sde] Write Protect is off
[335037.243192] sd 7:0:0:0: [sde] Mode Sense: 22 00 00 00
[335037.243403] sd 7:0:0:0: [sde] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[335037.274280] sd 7:0:0:0: [sde] Attached SCSI removable disk
```

This lead me to suspect that the communication with the MSC device is not really working in the composite.

alambe94, 2021-09-21:
Hi,
My observation
MCU 411CE
Total Endpoints 3 (Excluding EP0)
CDC requires 2 IN EP & 1 OUT
MSC requires 1 IN & 1 OUT 
at this point you have consumed all IN endpoint
Total PMA Buffer is 1.25KB (1280 Bytes)

HAL_PCDEx_SetTxFiFo takes size in WORDS, minimum is 16 words.
so total buffer consumption is
EP OUT Common (128*4) is 512 bytes
EP0 IN (64*4) 256

CDC IN1 (128*4) 512
CDC IN2 (64*4) 256
MSC IN (128*4)512

as above calculation suggest you have exhausted PMA buffer

my suggestion will be to partition PMA as

EP OUT Common (512 Bytes) 128 as param to HAL_PCDEx_SetTxFiFo()

EP0 IN (64 Bytes) 16
CDC IN1 (128 Bytes) 32
CDC IN2 (64 Bytes) 16
MSC IN (128 Bytes) 32

lbthomsen, 2021-09-21:
So - in the setup it is possible to add 2 CDC interfaces but how would that ever be possible - 2 IN EP's per device is 4 already + EP0 would be five.

I will try with HID + MSC only!

lbthomsen, 2021-09-21:
And yes - CDC + MSC only with the buffers you suggested works:
```
[356454.344290] usb 2-4.5.4: new full-speed USB device number 5 using xhci_hcd
[356454.446124] usb 2-4.5.4: New USB device found, idVendor=0483, idProduct=52a4, bcdDevice= 2.00
[356454.446126] usb 2-4.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[356454.446127] usb 2-4.5.4: Product: STM32 COMPOSITE DEVICE
[356454.446128] usb 2-4.5.4: Manufacturer: STMicroelectronics
[356454.446129] usb 2-4.5.4: SerialNumber: 318A34583030
[356454.458855] cdc_acm 2-4.5.4:1.0: ttyACM1: USB ACM device
[356454.459382] usb-storage 2-4.5.4:1.2: USB Mass Storage device detected
[356454.459566] scsi host7: usb-storage 2-4.5.4:1.2
[356455.460723] scsi 7:0:0:0: Direct-Access     STM      Product          0.01 PQ: 0 ANSI: 2
[356455.460947] sd 7:0:0:0: Attached scsi generic sg4 type 0
[356455.461339] sd 7:0:0:0: [sde] 200 512-byte logical blocks: (102 kB/100 KiB)
[356455.461589] sd 7:0:0:0: [sde] Write Protect is off
[356455.461591] sd 7:0:0:0: [sde] Mode Sense: 22 00 00 00
[356455.461819] sd 7:0:0:0: [sde] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[356455.495957] sd 7:0:0:0: [sde] Attached SCSI removable disk
```

closed 2021-10-10 by alambe94