alambe94/I-CUBE-USBD-Composite #2: User code is overwritten on CubeMX code generation

lbthomsen, 2021-09-22:
When generating code from CubeMX all source files are completely overwritten.  Even "user code" sections - like this (from `usbd_msc_if.c`:

```
/* USER CODE BEGIN PV */
/* Private variables ---------------------------------------------------------*/
uint8_t MSC_Storage[32*1024];
/* USER CODE END PV */
```

alambe94, 2021-10-06:
CubeMX Templates are not implemented yet.

lbthomsen, 2021-10-11:
Sure :) Let me know if you want me to test something when or if you implement it.

sammok, 2023-05-17:
for me, once you completed your config, you can try move AL94_USB_Composite/ to root directory out of Middlewares/ and make a copy of Composite/

then go to the IOC, cancel the I-CUBE-USBD-COMPOSITE

then go to project properties -> C/C++ General -> Paths and Symbols, add AL94_USB_Composite/ and Composite/ path to your compiler

lbthomsen, 2023-05-24:
Been doing the same pretty much