alambe94/I-CUBE-USBD-Composite #5: Enumeration in FULL SPEED not working

st0ne, 2021-12-17:
in usbd_core.c line 178

```
/* Get Device Configuration Descriptor */
  if (pdev->dev_speed == USBD_SPEED_HIGH)
```
here is a decision between HIGH SPEED and FULL SPEED but pdev->dev_speed is never set so it will be 0 and used as USBD_SPEED_HIGH. Therefore wrong descriptor settings will be used and enumeration is not working.

Fix could be to set dev_speed with #if (USBD_USE_HS == 1) somewhere before.

alambe94, 2021-12-17:
The core(hardware) speed is set with USBD_LL_SetSpeed() from PCD reset callback in usbs_conf.c
which is decided by hpcd->Init.speed configured in cube

USBD_Init() set the stack speed in usb_device.c

st0ne, 2021-12-17:
ok, but where is set dev_speed then? i cannot see it. 
If i debug the code it goes stright into highspeed configuration but "speed" for USBD_LL_SetSpeed is set with USBD_SPEED_FULL

if i set pdev->dev_speed manually to FULL SPEED then it works

alambe94, 2021-12-17:
usb.c or usb_otg.c which automatically generated by cube. In this we have function name MX_USB_XXXXXXX_PCD_Init
in this we set 
XXXXXXX.Init.speed = PCD_SPEED_HIGH Or PCD_SPEED_FULL

which will decide pdev->dev_speed USBD_SPEED_HIGH or USBD_SPEED_FULL
in PCD reset callback

st0ne, 2021-12-17:
```
void MX_USB_PCD_Init(void)
{

  /* USER CODE BEGIN USB_Init 0 */

  /* USER CODE END USB_Init 0 */

  /* USER CODE BEGIN USB_Init 1 */

  /* USER CODE END USB_Init 1 */
  hpcd_USB_FS.Instance = USB;
  hpcd_USB_FS.Init.dev_endpoints = 8;
  hpcd_USB_FS.Init.speed = PCD_SPEED_FULL;
  hpcd_USB_FS.Init.phy_itface = PCD_PHY_EMBEDDED;
  hpcd_USB_FS.Init.Sof_enable = ENABLE;
  hpcd_USB_FS.Init.low_power_enable = DISABLE;
  hpcd_USB_FS.Init.lpm_enable = ENABLE;
  hpcd_USB_FS.Init.battery_charging_enable = DISABLE;
  if (HAL_PCD_Init(&hpcd_USB_FS) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN USB_Init 2 */

  /* USER CODE END USB_Init 2 */

}
```

this is how it is set on my cpu. looks good. but for some reason somewhere it switches to highspeed.
i have also the problem that USB Device Tree Viewer in windows shows:

`      ========================== Summary =========================
Vendor ID                : 0x0483 (STMicroelectronics)
Product ID               : 0x52A4
USB Version              : 2.00 -> wrong, Device is Full-Speed only
Port maximum Speed       : High-Speed
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Self powered             : yes
Demanded Current         : 100 mA
Used Endpoints           : 7`

dont know where to change that and i have problems with audio devices. they are not activated from windows. maybe it has something to do with the version?

alambe94, 2021-12-17:
put breakpoint in HAL_PCD_ResetCallback()
single step and see what value USBD_LL_SetSpeed() this function gets

st0ne, 2021-12-17:
ok i found the "version". Its bcdUSB in USBD_DeviceDesc. 
I changed it to 0x00, 0x01 and now the Version Error is gone but Audio Enumeration is not working anyways, so this has to be something different

st0ne, 2021-12-17:
> put breakpoint in HAL_PCD_ResetCallback()
> single step and see what value USBD_LL_SetSpeed() this function gets

it is definitive USBD_FULL_SPEED

but if i do not set it manually again before

if (USBD_RegisterClass(&hUsbDevice, &USBD_COMPOSITE) != USBD_OK)

then it will be 0 again.

alambe94, 2021-12-17:
have you enabled only audio or other classes as well
because audio is not working with other classes except  HID
I have mentioned this bin doto

st0ne, 2021-12-17:
aaah, i enabled it with 2 CDC Ports. So why this is not working together?

alambe94, 2021-12-17:
I don't know
still figuring out. even tried putting cdc in the last

st0ne, 2021-12-17:
[image]

This is how it looks in windows. Maybe it is useful to open another issue here for this?

alambe94, 2021-12-17:
audio plus other class, descriptors captured with wire shark looks good but enumeration fails

alambe94, 2021-12-17:
> [image]
> 
> This is how it looks in windows. Maybe it is useful to open another issue here for this?

yes same issue, I don't know what is going on.
need to try on ubuntu

st0ne, 2021-12-17:
are there audio controls implemented? mute, volume and so on?

alambe94, 2021-12-18:
have not tried all features but the microphone implementation says it supports audio control
the speaker only support mute control

st0ne, 2021-12-18:
in Ubuntu it is detected as 2xACM device and for audio as STM32 Composite SPDIF and STM32 Analog Output
But no microphone is detected...
I will try if audio output is working

st0ne, 2021-12-18:
its strange. I added code to play back samples on I2S and now only microphone shows up in ubuntu and no speaker anymore...

alambe94, 2021-12-18:
are you using audacity?

st0ne, 2021-12-18:
[image]
directly in sound settings, but i can try audacity

alambe94, 2021-12-18:
are you able to enumerate in full speed mode?
or you still have to manually in FS mode

alambe94, 2021-12-18:
reset function will be called when you plug the USB, then it will set the speed accordingly.
You are saying that after this it is getting reset to high speed?