alambe94/I-CUBE-USBD-Composite #7: Not working on STM32L5

mrhodel, 2021-12-27:
The package works great on my STM32F746 Disco board. However, not so much on a Nucleo-L552.
The composite USB is configured for CDC+MSC and I ticked the STM32F1_DEVICE option. The L5 doesn't have OTG and won't compile without the F1 selection.

As soon as the USB is enabled, I expect to get a USB interrupt, but nothing. I've stepped through the code and the USB register setup seems okay.

Is there something special about the L5 that needs configuration?

alambe94, 2021-12-27:
for L5 I am seeing USBD_Clock_Restore() is called in resume callback which is not there in other families. Other than that, I am not seeing any difference.

mrhodel, 2021-12-27:
That was quick ;-), thanks.
I've run the L5 with CDC only and MSC only and they both enumerate individually.

I assume that you've run composite on an F1 or similar.
Any chance you can try it on an L5?

mrhodel, 2022-01-10:
I fixed this issue by adding the following line of code to main.c in before calling MX_USB_DEVICE_Init().

HAL_PWREx_EnableVddUSB()

closed 2022-01-15 by alambe94