alambe94/I-CUBE-USBD-Composite #8: CDC + AUDIO Composite not working together

st0ne, 2022-01-15:
i still working on getting CDC + AUDIO working.

so i found this:
https://community.nxp.com/t5/Kinetis-Software-Development-Kit/Composite-USB-CDC-and-Audio-Class/m-p/486885#M4306

alambe94, 2022-01-15:
Ok, will try with IAD descriptor

st0ne, 2022-01-16:
Currently i look at the descriptor table at the teensy project:
https://github.com/PaulStoffregen/cores/blob/ee3a0ada4af8338bc24f1cf12eb696dc6aa02b3b/teensy3/usb_desc.c

But be careful. Windows does not like a wrong descriptor. At the moment its crashing as soon i connect the usb cable!
On Linux there is no problem, it gives a wrong descriptor length of 0.

st0ne, 2022-03-15:
hi, found the issue now.

yes its the missing IAD. I added it to MIC and SPKR.
changed the size of descriptor of both +8 and also changed USBD_Update_Audio_SPKR_DESC and USBD_Update_Audio_MIC_DESC.

Windows is detecting audio devices correct but i have to test if everything is working. but looks good at the moment :-)

st0ne, 2022-03-22:
i also changed:

#if (USBD_USE_UAC_SPKR == 1)
    HAL_PCDEx_PMAConfig((PCD_HandleTypeDef *)pdev->pData, AUDIO_SPKR_EP, PCD_DBL_BUF, (pma_track + AUDIO_OUT_PACKET) | ((pma_track + AUDIO_OUT_PACKET) << 16));
    pma_track += ((128 + AUDIO_OUT_PACKET) * 2);
#endif

after that sound was fine.

but have problems with microphone. do not get it to work that samples are sent to usb endpoint...

st0ne, 2022-03-28:
mic needs extra endpoint. it conficts with EP1 from Speaker