alambe94/I-CUBE-USBD-Composite #9: 407 CDC +HID(keyb) + FS + Linux not working

fsartori66, 2022-01-25:
Just building straight from cube IDE.

Using  Full Speed device

Selected:
ACM (1)
HID Keyboard

Serial device /dev/ttyACM0 is not mounted 
Only keyboard:

dmesg
21199.438045] usb 1-3.4: new full-speed USB device number 35 using ehci-pci
[21199.548427] usb 1-3.4: New USB device found, idVendor=0483, idProduct=52a4, bcdDevice= 2.00
[21199.548430] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[21199.548432] usb 1-3.4: Product: STM32 COMPOSITE DEVICE
[21199.548433] usb 1-3.4: Manufacturer: STMicroelectronics
[21199.548434] usb 1-3.4: SerialNumber: 3659326D3431
[21199.551630] input: STMicroelectronics STM32 COMPOSITE DEVICE Keyboard as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3.4/1-3.4:1.0/0003:0483:52A4.0004/input/input17
[21199.614196] input: STMicroelectronics STM32 COMPOSITE DEVICE Consumer Control as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3.4/1-3.4:1.0/0003:0483:52A4.0004/input/input18
[21199.614260] input: STMicroelectronics STM32 COMPOSITE DEVICE System Control as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3.4/1-3.4:1.0/0003:0483:52A4.0004/input/input19
[21199.614367] hid-generic 0003:0483:52A4.0004: input,hiddev0,hidraw3: USB HID v1.11 Keyboard [STMicroelectronics STM32 COMPOSITE DEVICE] on usb-0000:00:12.2-3.4/input0

linux lsusb -v

Bus 001 Device 035: ID 0483:52a4 STMicroelectronics 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x52a4 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0064
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              6 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     187
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         1
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      7 
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          2
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        1
        bSlaveInterface         2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
fisa@fisa-desktop:~$ 

if I enable only the  ACM then I get it mounted as /dev/ttyACM0

This device has 
1 bidirectional control endpoint0
• 3 IN endpoints (EPs) configurable to support Bulk, Interrupt or Isochronous transfers
• 3 OUT endpoints configurable to support Bulk, Interrupt or Isochronous transfer

Is that the problem? Lack of end points?
How do I verify this?
I thought that ACM used 2 out 1 in and keyb 1 and 1  ??

Thanks
And a nice job!

fsartori66, 2022-01-28:
looks like this is a linux issue
on windows10 it works

fsartori66, 2022-01-28:
I was able to get it working:
File: usb_cdc_acm.c
2170     desc[16] = str_idx;

Before was overwriting position 15 which correspond to the bInterfaceProtocol

alambe94, 2022-01-28:
That's great.
earlier acm string descriptor was at 7 (IAD)

will update to 16
thanks

closed 2022-01-28 by fsartori66