alambe94/I-CUBE-USBD-Composite #11: Support for L4 non-OTG devices

jacky4566, 2022-02-11:
Does this pack support L4 non-OTG devices?

Trying to compile with STM32L433CC and getting errors about OTG

`../Middlewares/Third_Party/AL94_USB_Composite/COMPOSITE/Target/usbd_conf.c:31:10: fatal error: usb_otg.h: No such file or directory`

alambe94, 2022-02-12:
Tick F1 device and generate cube code as pair of .c/.h

ucahmetuc, 2024-01-03:
> Tick F1 device and generate cube code as pair of .c/.h

This method works on STM32L053C8T6. But no matter what I did, it didn't work on STM32L433VCT6.

ucahmetuc, 2024-01-03:
I solved the problem. Add "**HAL_PWREx_EnableVddUSB();**" to the top of the "MX_USB_PCD_Init" function. When you added the line, it started working smoothly.