alambe94/I-CUBE-USBD-Composite #17: The lib don't work for HID, CDC and HID+CDC

JulienGuitter, 2022-05-26:
Hello, I have a project that need CDC and HID. But i can't make work this lib. I have try separately CDC or HID, and i can't, make it work. When i plug usb, the computer said, there are error in the chip. But if i use the lib of stm32cube, it works. Can you help me please. 
Thanks in advance

alambe94, 2022-06-01:
what is your MCU

JulienGuitter, 2022-06-01:
And if ibhave to modificate dôme value, i don't know wich

alambe94, 2022-06-01:
Is the MCU clock configured correctly?
External Crystal and 48MHz or cube will recalculate if USB is enabled

Is USB interrupt enabled in NVIC?

Did you call MX_USB_DEVICE_Init()?

alambe94, 2022-06-01:
is going hard fault after executing MX_USB_DEVICE_Init()?
is usb interrupts hitting?

JulienGuitter, 2022-06-01:
So i find my probleme, i juste forgot to check stm32f1 in conf, so now it's work but i don't know how to use HID, i past my code but it didn't work.

JulienGuitter, 2022-06-01:
So i finaly did it, i have change the hid report desc for the basic and it's work. Thank's for your help

closed 2022-07-03 by alambe94