alambe94/I-CUBE-USBD-Composite #21: CDC_ACM + Hid keyboard issue

nckpln, 2022-10-08:
My setup:
MCU stm32f103c8
CubeMX v.6.3.0
AL94.I-CUBE-USBD-COMPOSITE v.1.0.3

app.c

```
#include "main.h"
#include "usbd_def.h"

extern USBD_HandleTypeDef hUsbDevice;

uint8_t kbdPress[8] = {0,0,4,0,0,0,0,0};
uint8_t kbdRelease[8] = {0,0,0,0,0,0,0,0};

#define LED_ON() LL_GPIO_ResetOutputPin(LED_GPIO_Port, LED_Pin)
#define LED_OFF() LL_GPIO_SetOutputPin(LED_GPIO_Port, LED_Pin)
#define LED_TOGGLE() LL_GPIO_TogglePin(LED_GPIO_Port, LED_Pin)
#define GET_PTT() LL_GPIO_IsInputPinSet(PTT_GPIO_Port, PTT_Pin)

void print(char *str)
{
    uint16_t c = 0;
    while (str[c] != '\0')
        c++;
    CDC_Transmit(0, (uint8_t *) str, c);
}

void KeyDown(void)
{

    uint8_t res = USBD_HID_Keybaord_SendReport(&hUsbDevice, kbdPress, 8);
    if(res)
        print("KEY DOWN FAIL\r\n");
    else
        print("KEY DOWN OK\r\n");        
}

void KeyUp(void)
{
    uint8_t res = USBD_HID_Keybaord_SendReport(&hUsbDevice, kbdRelease, 8);
    if(res)
        print("KEY UP FAIL\r\n");
    else
        print("KEY UP OK\r\n"); 
}

void app_init(void)
{
    LED_OFF();
}

void app_loop(void)
{
    KeyDown();
    LED_ON();
    LL_mDelay(1000);
    KeyUp();
    LED_OFF();
    LL_mDelay(3000);
}
```

LED blinks, serial terminal prints "KEY DOWN OK/KEY UP OK" but no input comes from keyboard
Issues #17 and #19 are kind of similar to mine, but both them resolved without clear explanation.

pitman75, 2022-10-23:
More information about my changes for HID keyboard added.

andolon, 2023-01-18:
@nckpln  : It was **not** working for me. So, I used a working  Keyboard Report Descriptor extracted from here: 
https://www.instructables.com/STM32-As-HID-USB-Keyboard-STM32-Tutorials/
and it works.   Receiving chars.